chirimen-oh/chirimen.org #5: DC ファン配線図をよりミスしにくいものにしたい

dynamis, 2018-08-12:
https://github.com/chirimen-oh/tutorials/blob/master/RaspberryPi/JA/imgs/section1/DC3.png

* 10k と 1k の抵抗がディスプレイの発色が良いものを使わないと見間違える学生が出てくることが多い
  * よく見ろと言ってもミスするので、抵抗だけ大きくズームして配色の判別がつくようにしたい
* 黒いジャンパーピンを内側の GND ピンではなく外側の GPIO 21 番ピンに繋ぐ学生がでてくる
  * 回路図やコードから見ても明らかなんだけど、図として見にくいのは確か。黒いジャンパーピンは右からではなく下から線を引いたりした方が見やすいかも知れない

satakagi, 2018-08-20:
実態配線図のソースデータも共有した方が良いですね。

tadfmac, 2018-08-21:
すごい切り貼りで作ってるんですよね。。。
Sketch.app のデータなので、Macでしか編集できないです。

dynamis, 2018-08-27:
この手のもののための標準ツールがあって使っているわけじゃなかったのですね。
fritzing とかだと思っていたら Sketch.app のデータですか。

https://www.sketchapp.com/pricing/ だと有料アプリなのでちょっと辛いですね

ちなみにサイズの小さなファイルなら issue のコメント記入欄にドロップするだけでもアップロード (＆リンク) できたりします。

tadfmac, 2018-08-29:
> https://www.sketchapp.com/pricing/ だと有料アプリなのでちょっと辛いですね

..... 本コミュニティでは、別repoにeagleのデータもHWデザインとして共有している時点で、特に有料アプリだと辛いという感覚はないですが.....

satakagi, 2018-09-03:
ベースの絵はfritzingで作っていると思うのですが、できるだけfritzingで作るという方針は良いですよね？

satakagi, 2018-09-03:
テスターを使った抵抗の測定方法、0章に外部サイトへのリンクを張りました。

satakagi, 2018-09-03:
[image]
このぐらいまではFritzingだけで編集できますが、これで十分ではないでしょうか？

closed 2018-09-04 by satakagi