chirimen-oh/chirimen.org #14: インデックスページからチュートリアルへのリンクgithub pages経由でリンク切れしてる

satakagi, 2018-10-04:
chirimen.org/tutorial/
でリンク切れです。
理由は、github pagesはcase sentiiveかつ拡張子も必要のため

github pages止めるか、pagesでもリンク切れないようにcaseを合わせるかどちらか

satakagi, 2018-10-04:
多分今はあまり意味のない
/RaspberryPi/JA/
こっちのページをリンク切れないように直してみました。
このページがtutorials.chirimen.orgからでも問題ないなら、
/
のほうのページの中身も同じように修正するか　github pagesを廃止するかどちらかですね。

satakagi, 2018-10-04:
https://tutorial.chirimen.org/raspberrypi/ja/
で試しにアクセスしてみたら、
.md拡張子がついているとうまくないですね。
どうしますか？ @dynamis

dynamis, 2018-10-11:
こちら返事できていなくてごめんなさい (口頭では少し話したと思いますが)。
github pages ではなく netlify で使う方を本番サイトとする想定をしています

基本的には tutorial.chirimen.org 配下、.html とか .md とかの拡張子無しのシンプルかつ小文字に統一の URL に統一するのが見た目も綺麗だし大文字小文字間違えても勝手にリダイレクトされるし、ユーザ目線で良いと考えています。

github pages のほうは誤解がないように無効化しておきます。

リンクするときはそういう URL になると言うことだけ考えて、github pages の時と同じように markdown ファイルを書き換えていってもらえれば大丈夫な、Jekyll テンプレートをカスタマイズしたからといって編集者が分からなくなったり難しくなったりする要素は極力排除したつもりです。

closed 2018-10-11 by dynamis