chirimen-oh/chirimen.org #16: Feature Policy を踏まえた説明に更新する

dynamis, 2018-10-10:
Web Bluetooth などは Chrome 64 以降での Feature Policy の制限により iframe を使う jsbin では動かなくなっているはず。jsfiddle のようにサービス側が明示許可するようにしてくれない限り使えないので、そのあたり説明を入れる必要がある。

該当する機能:
- [ ] Web Bluetooth

dynamis, 2019-03-27:
- [ ] google や mozilla (MDN) などで feature policy の説明が書いているページを探す
- [ ] bluetooth 使う説明を書いているところに、jsbin 等では動かないからローカル HTML か jsfiddle (実際に動作確認してから) で試してください等と書き足す

saxAllan, 2019-04-01:
feature policy の説明：
- https://w3c.github.io/webappsec-feature-policy/
- https://developers.google.com/web/updates/2018/06/feature-policy/

dynamis, 2019-04-09:
こちらのコミットですね https://github.com/chirimen-oh/tutorials/commit/6ee54fd24c4777de1cfbf55cb8e7bd1950972531

> Note: これらのAPIは Feature Policy の制限（iframe 内からの使用禁止）により、jsbin では動作しません。ですから、ローカルで実行するか jsfiddle 等の各種 API が利用可能としているサービスを利用して実行してください。 ※なお、jsfiddle では実行可ということになっていますが、上手く動作しない例もあるようです。必ず自身で動作することを確認してから利用するようにしてください。

- [ ] Feature Policy を 「iframe 内からの使用禁止」と書くと語弊があります。「iframe 内からは埋め込み側で明示許可されていない API は利用不可」です。説明は正確に書くように注意してください。

- [ ]  動作確認するのは読者じゃなくて執筆側の仕事という意図でした。「動作確認して使ってね」と書いて欲しいのではなく、動作確認をしてから「ここなら確認済みで大丈夫だよ」って書いて欲しいのです。

saxAllan, 2019-04-11:
「iframe 内からは埋め込み側で明示許可されていない API は利用不可」に修正しました。
まだ動作確認ができていない為 jsfiddle の紹介をし、動作検証がまだできておらず近日中に行う旨を追記しました。

dynamis, 2019-04-18:
ありがとうございます。jsfiddle で大丈夫と再確認できたら「まだ動作確認ができていません。近日中に動作確認を行う予定です。」などの記述を修正しましょう

https://tutorial.chirimen.org/raspi3/ja/section4

saxAllan, 2020-02-12:
JSFiddle では今現在利用できないことを確認したため、利用可能である記述を削除しました。
代わりに、CSBやローカルでの実行をお勧めする記述をしました。

closed 2020-02-12 by saxAllan