chirimen-oh/chirimen.org #23: Google 翻訳でコードブロックの中身が壊れる

dynamis, 2018-10-28:
当たり前といえば当たり前なのだが、チュートリアルを Gooogle 翻訳で英語にして読んでいる場合にコードブロックをそのままコピペして実行しようとすると動かないことがある。

全て英語でちゃんと書かれたページを用意できればそれで良いのだけど、何か他に対策がないか。

 - 翻訳対象外とマークする方法がないか
  - lang=en とか付けておけばスルーしたりしないか？

dynamis, 2018-10-31:
class="notranslate" を付ければ良いらしい

https://webmasters.googleblog.com/2008/10/helping-you-break-language-barrier.html

ただコードブロックを書く度にこれを付けるのって超面倒で絶対作業抜けが発生する。
サイトのビルド (jekyll や netlify) 側で何とか出来ないか検討したいところ。

sizuhiko, 2018-11-01:
現時点のGitHub PagesはJekyll3系なので、以下のように `_config.yml` に記述すると `class` を追加できると思います。

```yml
kramdown:
  input: GFM
  syntax_highlighter: rouge
  syntax_highlighter_opts:
    css_class: 'notranslate'
```

dynamis, 2019-10-16:
折角コメント頂いていたのに見逃してましたごめんなさい。 syntax_highlighter_opts とかあったのですね。ただ、GitHub ドキュメントで書かれている kramdown の設定しててもどうも GitHub Pages と挙動違うなと調べた結果、CommonMarkGhPages 使うようになってた https://github.com/github/pages-gem/releases/tag/v172 ためそれに合わせています。そちらで使ってる syntax highight 周りのオプション指定を調べないとですね。

あと、付加する属性としては translate グローバル属性 `<p translate="no">` の方良さそうでした。任意属性が付与できないなら class で良いけど。