chirimen-oh/chirimen.org #36: 短い hello real world ページを用意する

dynamis, 2019-03-11:
よく分からなくとも、取りあえずこうすれば L チカやセンサーの読み取りが可能という簡単なはろーわーるどページ (CHIRIMEN 的には伝統的に？ Hello Read World) を用意する。

現在作成してもらっている途中のページがこちら:

https://tutorial.chirimen.org/raspi3/ja/helloworld
https://tutorial.chirimen.org/ty51822r3/ja/helloworld

dynamis, 2019-03-11:
気になっている点

* 短いので目次は要らなさそう
* 実行用にローカルファイルのパス以外にオンラインのページへのリンクを付ける
  * github pages と jsbin へのリンクを両方。Raspi 上のファイル 〜 やオンライン版で確認したり、jsbin 版でコードを書き換えて試して頂けます、程度にローカル以外へのリンクも。
  * jsbin リンクは (リンク先が変わっても良いようにリダイレクトリンクで https://r.chirimen.org/gpio-blink など)
* まとめは多分要らない。各セクションの中でその他の例へのリンクを示した方が良いかも
  * その他の GPIO の例は http://chirimen.org/chirimen-raspi3/gc/top/examples/#gpioExamples
  * その他の I2C の例は http://chirimen.org/chirimen-raspi3/gc/top/examples/#i2cExamples
  * 応用例はこちら http://chirimen.org/chirimen-raspi3/gc/top/examples/#advanced
* コードサンプルは本文に書かないと配線するだけでコードを学ばない感じなので、コードは一応含めておいた方がよさそう。

詳しい説明はないが、慣れた人でも何度も使うことになるだろうページ (examples など) にはリンクしてあげる感じで。

TakumiNishina, 2019-03-12:
jsbinのリンク張り（リダイレクトリンク）
まとめ、目次の削除を実行しました。

コードは、張り付けることスペースをとるので、raspiは、JSbin、TY51822r3は、Githubのリンクを張りました。

TakumiNishina, 2019-03-12:
raspyと、ty51822両方ともHelloRealWorld.md に変更しました。

各チュートリアルの top page にリンクをはりました。

dynamis, 2019-03-22:
現状まだ問題が多いので reopen します。全体的に、簡潔だが正確かつ初心者にも慣れてる人にも読みやすいように更新してください:

https://tutorial.chirimen.org/raspi3/ja/hellorealworld

> 用意するもの

なるべく入手先の例にリンクしてください。

> 右図と同じように配線してみよう。

html 開いているスクリーンショットはあるが、回路図に該当する図はないです。説明は正確に。

> LED には方向がある。アノードが足が長い方。反対の足が短い方をカソードと言い GND 側に繋ぐ。抵抗はどちらかに繋ぐ。

もう少し親切な表現で:

LED には極性 (方向) があり、足が長い方 (アノード) を GPIO 出力ピンに、足が短い方 (カソード) を GND 側に繋いでください。抵抗は LED のどちらの足の側に繋いでも構いません。

> /home/pi/Desktop/gc/gpio/LEDblink/index.html をダブルクリックすると ...
本当の初心者さんにはこれでは分からないので次のような感じで:

`/home/pi/Desktop/gc/gpio/LEDblink/index.html` ファイル (デスクトップの gc ディレクトリを開き、gpio, LEDblink ディレクトリを開くとあります) をダブルクリックで開くと ...

> オンライン Example はこちら
> 今回使用したコードはこちら

両方同じリンクになってます。

> その他の GPIO の例はこちら

その他の GPIO 利用例 (スイッチのオンオフや人感センサーを使うなど) はこちら

くらい書いてあげた方が良いと思います。

> L チカを行った回路
> 右図に書いてあるもの

Lチカ回路は使わないし、右に図はなく写真の中の文字は小さすぎて読みにくいです。単純に

- ジャンパーワイヤー (メス-メス) x 4
- ADT7410使用 I2C 温度センサーモジュール

だけ書き、入手先にリンクすべきかと思います。

> 右図のような画面が表示されるはずです。

i2cdetect が何をしているのか説明がないのでちゃんと書く

> 右下に数字がでていますね。

ブラウザの画面右下じゃなくて中央下だと思います。

dynamis, 2019-07-24:
その後色々修正して完成したと思うのでクローズ

closed 2019-07-24 by dynamis