chirimen-oh/chirimen.org #38: micro:bit の説明ページを統合・用意する

dynamis, 2019-03-22:
現在 micro:bit 版は独立した github pages にあり、raspi3 のチュートリアルの発展編 -  5. WebBluetooth編 からリンクされているだけの状態:

https://tutorial.chirimen.org/raspi3/ja/
http://chirimen.org/webGPIO-etc-on-microbit-via-webBluetooth/

これを tutorial.chirimen.org 配下に統合した方が良い気がする。raspi 3 との接続を前提とした説明だけなら https://tutorial.chirimen.org/raspi3/ja/ 配下ですが、それだけでも無さそうなので ty ボードと同様に

https://tutorial.chirimen.org/microbit/ja/

配下でしょうか。ただ、現状のドキュメントは 1 枚にまとまっている代わりに初心者向け情報にななっていなので、それに加えて、LED 1 つだけを GPIO で L チカしたりスイッチで点滅するだけのサンプルコードと説明ページ、短い hellorealwold ページを用意した方がよさそう。

satakagi, 2019-03-25:
WebI2C/micro:bitの実装中（ただいろいろ難航中）で、これが動くようになったところで、CHIRIMEN with micro:bitに名称変更をする予定です。
また、この実装に伴って、使い方(APIのスペック含め)がいろいろ変更することになると思います。もし急ぎの用事がないようでしたら、この実装のめどがついてからtutorialsの調整をしませんか？
目途としては・・梅雨ごろまでにはｗ

dynamis, 2019-03-25:
承知しました。それでは micro:bit 説明の統合とかは satakagi さんから OK ですって指示を頂くまで pending しましょうか。

dynamis, 2019-07-24:
raspi3 のメインチュートリアルがそのまま同じように使える環境なので、違いだけが書かれた独立ドキュメントサイトで進める方向に。統合ではなく上手く紹介を入れる形にする

現在こちら: http://chirimen.org/chirimen-micro-bit/

dynamis, 2019-08-17:
サイトのヘッダ、トップなどからの誘導は完了。残りは...

- [ ] Raspi3 チュートリアルの発展編からの紹介を最新のものに
- [ ] リモート制御したいデバイスがある時やスマホと連携したいときにはオススメとか何処かに記述。例えばスマホで位置情報を得ながら操作するようなデバイスは micro:bit で

dynamis, 2020-06-17:
こちらはクローズして #75 にて引き継ぐ

closed 2020-06-17 by dynamis