chirimen-oh/chirimen.org #41: スターターキット変更に合わせて基礎編を更新

dynamis, 2019-04-21:
スターターキットの中身を変更 #40 するのが確定したらそれに合わせてチュートリアルを書き換える。

- [ ] 旧チュートリアル向けのページはバックアップとして別ページに残す
- [ ] 旧チュートリアル向けの案内を追加
- [ ] 新しい部材を前提としたものに書き換えを進める
  - タクトスイッチ
  - ジャンパーピン
  - モーター

dynamis, 2019-06-12:
変更前のものは webiot-2018 branch 切った:
https://github.com/chirimen-oh/tutorials/tree/webiot-2018

2018 年度版のスターターキットで取り組みたい人はブランチデプロイしたサイトを見れば良い https://webiot-2018--tutorial-chirimen-org.netlify.com/

satakagi, 2019-06-24:
ブレッドボードの図：
https://github.com/chirimen-oh/tutorials/blob/master/raspi3/ja/imgs/section1/DC3motor.png
そのソースコード（ちびギアモータのパーツを新たに制作）
https://github.com/chirimen-oh/tutorials/blob/master/raspi3/ja/imgs/section1/DC3motor.fzz

Section1の該当部分はコメント行として上記ブレッドボード図を書き込んであります：
https://github.com/chirimen-oh/tutorials/blob/master/raspi3/ja/section1.md

dynamis, 2019-07-22:
チェックリスト漏れ。section 4 でも DC ファンが出てきてるので置き換えが必要
- [ ] https://tutorial.chirimen.org/raspi3/ja/section4

dynamis, 2019-09-25:
section5 の変更は結構大変そう。発展編だし特に推奨センサーじゃなく GP2Y0E03 無くても良いからそのままにするか、VL53L0X or GP2Y0E03 どちらかを使って下さいくらいとして説明文とコードだけ直すのでも良いかも。

dynamis, 2019-10-01:
- [ ] デスクトップスクリーンショット古い
  - https://tutorial.chirimen.org/raspi3/imgs/section0/CHIRIMENforRaspberryPi3desktop.png

dynamis, 2019-12-02:
section5 はスターターキットを使ったものではないし対象外としてクローズ

closed 2019-12-02 by dynamis