chirimen-oh/chirimen.org #43: Markdown で Import Code Snippet したい

dynamis, 2019-05-17:
Github 同様の markdown ということで GFM を使っていて今のところ出来ないが、サンプルコードとチュートリアル本文の両方に同じコードを書いてメンテする必要が生じているのは望ましくないので js ファイルを markdown にインポートしたい。

例えば vuepress (markdown-it を利用) でこんな感じで出来るように

https://vuepress.vuejs.org/guide/markdown.html#import-code-snippets

GFM の拡張なのか Jekyll 側の処理なのか何かで出来ると良いなという願望メモ

kou029w, 2019-05-17:
Gistは [jekyll-gist](https://github.com/jekyll/jekyll-gist) で埋め込めるようです
gist-it.appspot.com は簡単そうですがXSSなどリスクあるかも

dynamis, 2019-07-23:
include は _includes ディレクトリ配下のファイルだけだし include_relative で挿入するのは配下のディレクトリのファイル限定なので今のディレクトリ構造では使えないんですよね。。。チュートリアル用のコードは同一リポジトリ内にまとめておきたいし。

> Note that you cannot use the ../ syntax to specify an include location that refers to a higher-level directory.

dynamis, 2019-07-24:
ディレクトリ名規則が /targetboard/locale/... になってるのを /locale/targetboard/... にすれば当然 include_relative が使えるし、そのようにすれば根性で複数 locale ディレクトリ内のファイルを編集するのではなく Polyglot での i18n 対応が可能 (実際のファイルを locale ディレクトリに入れずに並べておける＆デフォルト言語へのフォールバックが出来る＆相対リンクのパスとかの処理が出来る)。

see also: https://github.com/chirimen-oh/chirimen-oh.github.io/issues/1

従って、次のように対応するのがよさそう:

- [ ] polyglot 導入
- [ ] /locale/targetboard/ パスに変更
  - ja ファイルは locale 省略なので /targetboard/pagename.md に
  - en ファイルは /targetboard/pagename-en.md にして front matter に title/lang/permalink を設定
  - en ファイルなどがある場合は ja ファイルも front matter を設定する (翻訳されないページは front matter ナシのままでもよい)
- [ ] 旧 /targetboard/local/ からはリダイレクトを設定 (netlify でよかった！)
- [ ] include_relative を使い始める
  - /targetboard/pagename.md から /targetboard/example/... にはアクセス可能に

dynamis, 2019-07-24:
polyglot 導入した https://github.com/chirimen-oh/tutorials/commit/26fe62dbf00b5c13e2a4f8ff478ff1c1ad820333

dynamis, 2019-07-24:
import 導入の進捗:

- [x] section0
- [ ] section1
- [ ] section2
- [ ] section3
- [ ] section4
- [ ] section5
- [ ] section6

note: 現在 jsbin でコードを書く想定になっているので js/css 読み込みタグの有無が異なるため HTML は include してはならないことに注意！

dynamis, 2019-08-15:
同一コードがあるものは include した。

closed 2019-08-15 by dynamis