chirimen-oh/chirimen.org #50: npm 移行に伴う変更

kou029w, 2019-07-31:
npm 移行に伴い (https://github.com/chirimen-oh/chirimen-raspi3/issues/68) `npm i @chirimen-raspi/polyfill` や https://unpkg.com/@chirimen-raspi/polyfill@1.0.1/polyfill.js などを用いることが可能になったが、一方で、既存の gc/drivers ディレクトリなどへの相対パスが機能しなくなったため、修正する必要がある

`/polyfill/` の文言が出現する文書

```sh
_redirects
raspi3/examples/section0/s0.html
raspi3/examples/section1/s1_1.html
raspi3/examples/section1/s1_2.html
raspi3/examples/section1/s1_3.html
raspi3/examples/section1/s1_4.html
raspi3/examples/section1/s1_5.html
raspi3/examples/section2/s2_1.html
raspi3/examples/section2/s2_2.html
raspi3/section0.md
raspi3/section1.md
raspi3/section2.md
raspi3/section3.md
raspi3/section4.md
ty51822r3/example/section0/s0.html
ty51822r3/example/section1/LEDblink/index.html
ty51822r3/example/section1/MultiBlinkAll/index.html
ty51822r3/example/section1/button/index.html
ty51822r3/example/section1/buttonAll/index.html
ty51822r3/example/section1/s1_1.html
ty51822r3/example/section1/s1_2.html
ty51822r3/example/section1/s1_3.html
ty51822r3/example/section1/s1_4.html
ty51822r3/example/section1/s1_5.html
ty51822r3/example/section2/s2_1.html
ty51822r3/example/section2/s2_2.html
ty51822r3/ja/diff.md
ty51822r3/ja/section0.md
ty51822r3/ja/section1.md
ty51822r3/ja/section2.md
ty51822r3/ja/section3.md
```

`/drivers/` の文言が出現する文書

```sh
_redirects
raspi3/examples/section2/s2_1.html
raspi3/section2.md
raspi3/section3.md
raspi3/section4.md
ty51822r3/example/section2/s2_1.html
ty51822r3/ja/diff.md
ty51822r3/ja/section2.md
ty51822r3/ja/section3.md
```

dynamis, 2019-08-09:
- [ ] 上記ページの修正の他に jsbin で drivers を参照している example が全滅しているので修正作業中

dynamis, 2019-08-10:
drivers が移動したがまた移動する可能性も備えて (今のところチュートリアルはオンライン参照のみなので) r.chirimen.org/xxxxx.js にした。その結果 tutorial 内では main.js 以外の相対パスでの polyfill/drivers 読み込みはなくなるので html の import もできるようになったはず

- [ ] html コードも本文にインポートさせて二重管理を避ける

dynamis, 2019-08-10:
- [x] /polyfill/ を含む raspi3 ページ
 - [x] /drivers/ を含む raspi3 ページ

同じように直して良いのか未確認なので保留
 - [ ] /polyfill/ を含む ty51822r3 ページ
 - [ ] /drivers/ を含む ty51822r3 ページ

dynamis, 2019-08-15:
driver パス変更の影響は対応したつもり。コーディングルール変更 #54 などが残っていることには注意。

closed 2019-08-15 by dynamis