chirimen-oh/chirimen.org #54: コーディングルールを gc 配下の example に合わせる

dynamis, 2019-08-15:
example とのコーディングルール統一として gc 側の修正を行った
https://github.com/chirimen-oh/chirimen-raspi3/issues/21#issuecomment-517090705

が合わせるだけでなく先行して実行タイミング制御 (onload から deferに) や mainFunction などについての扱いが変わっている。

tutorial 側もそれに合わせたい。

- [ ] 実行タイミング制御で onload 使わないで defer にする
  - [ ] script defer に関する説明を追加
- [ ] 関数名の変更 mainFunction から main に

kou029w, 2020-09-26:
Markdownコードブロックにコピペするのは簡単ですがメンテナンスし続けるのが困難なので、ページ読み込みは遅くなるのですが [emgithub.com](https://emgithub.com) や CodeSandbox の埋め込みをつかってリンクさせるのが良さそうに思いました。

dynamis, 2020-09-26:
example のコードを挿入するだけなら github のファイルを読み込んで埋め込む系のもので良いのですが、チュートリアルはその途中とか一部だけを抜き出したコードサンプルが大量にあるので example コードを参照させる形では解決不能ではないかと。

kou029w, 2020-09-26:
> 行数指定だと元コードが変わると困る

それは適宜tagやbranchを指定しておくかcommit idを指定しておけば解決しませんか?
例: https://emgithub.com/chirimen-oh/chirimen/blob/8a3c6f438be396c0388dcf04ab2281c54645ae11/gc/gpio/LEDblink/main.js#L11-L11

dynamis, 2020-09-26:
あ、はい。そうでした。example コードに存在しないコードというのもあるけど少数派だと思えばそこだけ諦めて Markdown 中に書くか、example 内に一緒に入れておいて同じフォーマットで参照しやすくするかというのは良さそうですね。

kou029w, 2020-11-20:
section2 に関しては https://emgithub.com/ を埋め込んでみています。問題なさそうであればコードの整形・コーディング規約の変更とともに引き続き切り替えていくつもりです