chirimen-oh/chirimen.org #56: partslist ページの内容拡充

dynamis, 2019-09-25:
悟さんがパーツリストを追加してくれたので、より見やすくしていきたい

https://tutorial.chirimen.org/raspi3/partslist

- [ ] 製品情報・仕様のページへのリンク
- [ ] 購入先例へのリンク
- [ ] できること、用途の簡単な説明を追加
- [ ] 抵抗も初学者向けの大きい (抵抗値が見やすい) 購入先を紹介
- [ ] netlify の url ルールに合わせて修正 (小文字ファイル名.md へリンク)

saxAllan, 2019-10-17:
ひとまず、「I2C センサー」の項に販売ページへのリンクを用意し、Examplesへのリンクを独立させておきました。販売ページ内に製品詳細が記載されている販売先を選んだので、1つ目と2つ目のチェックリストは併合してしまって良い気もします。

saxAllan, 2019-10-23:
各種リンクの列を作り、製品に関する情報へのリンクをまとめました。

kou029w, 2020-09-25:
[partslistページ](https://tutorial.chirimen.org/raspi/partslist.html)のデータがあまりにも膨大だったので[_data/partslist.csv](https://github.com/chirimen-oh/tutorials/blob/master/_data/partslist.csv)にCSVファイルとして外部化しました

kou029w, 2020-11-20:
これらは別課題として起票後に本Issueはクローズしたいと思います

> - [ ] 同じセンサーで違う見た目のパーツは写真の例を追加
> - [ ] netlify の url ルールに合わせて修正 (小文字ファイル名.md へリンク)

kou029w, 2020-11-20:
> - [ ] netlify の url ルールに合わせて修正 (小文字ファイル名.md へリンク)

よく分からないが少なくともREADME.md以外ファイル名に[A-Z]を含むファイルは存在しないように見えるのでクローズ。

closed 2020-11-20 by kou029w