chirimen-oh/chirimen.org #60: 標準温度センサーを ADT7410 から SH30 に切り替える

dynamis, 2019-10-10:
以下の理由により、チュートリアルで扱うセンサーを ATD7410 から SH30 に切り替えるというのは良いかもしれない。スターターキット入れ替え問題が発生するので今すぐにではありませんが。

現在ドライバーがシンプルに書ける (から WebI2C API の説明も出来る) という理由から ATD7410 を標準採用しているが、I2C センサーの中では少し高め (500円) かつ入手性としては実質秋月で買う 1 択。

温度センサーの計測範囲と精度比較についてはこちらの記事にあるが、ATD7410 に比べて SH30 が極端に劣るところはなく、オススメ温度センサーとされている:
https://lang-ship.com/blog/?p=502

ドライバーのコードはシンプルで湿度も同時に得られて価格は半額程度で Amazon/Ali から購入可能というのは魅力的な選択肢。
driver: https://chirimen.org/chirimen-raspi3/gc/contrib/examples/i2c-SHT30/node_modules/@chirimen-raspi/chirimen-driver-i2c-sht30/SHT30.js

同じ意味では BMP280 などもあるが Bosh のこのシリーズは気圧計と湿度計の補正で使うための温度計なので外部影響を受けやすく誤差も大きめとのことだし、ドライバーのコードも短くない。

dynamis, 2020-05-27:
昨年の ADT7410 セットを引き続き利用しているユーザもいるので、今年は ADT7410 から差し換えではなく ADT7410 / SHT30 の両対応ドキュメントにしたい。

kou029w, 2020-11-18:
https://chirimen.org/chirimen/gc/top/examples の上の方に移動しておく。CodeSandboxへのリンクを加える。

PR #81 (2020-11-18): SHT30 対応
resolved #60
closed 2020-11-18 by kou029w