chirimen-oh/chirimen.org #61: メインチュートリアルから Grove を外す (Grove 編に分離)

dynamis, 2019-10-10:
本チュートリアルのための機材は安価に揃えられるようにした方が CHIRIMEN 利用がしやすいが、チュートリアルに出てくる中で Grove センサー類が際立って高コストとなっている。

Grove も簡単で便利で良い場面はあるが、ジャンパー配線を先に教えている状態で Grove コネクタに degrade する意味はあまりないし、I2C Hub もブレッドボードでできるよって教える方が I2C に対する理解が確実に深まる。

ジャンパワイヤ使わずにワークショップしたい人向けの Grove 編として分離・残しつつ、メインのチュートリアルは全てジャンパワイヤ配線・安価で入手可能なセンサーに置き換えてしまうのが良いのではないか。

- Grove Light -> BH1750
- Grove Accelerometer -> MPU6050
- Grove Hub -> ジャンパワイヤとブレッドボード

dynamis, 2019-12-02:
https://tutorial.chirimen.org/raspi/section3

Phase 1: 脱 Grove

- [ ] Grove Light -> BH1750
- [ ] Grove Accelerometer -> MPU6050
- [ ] 複数接続を Grove Hub を使わない説明をメイン、Grove Hub も紹介する程度に
- [ ] ブレッドボードを使うときの機材のリストアップと写真の用意
  - 小さいブレッドボード x1, ジャンパワイヤ オス-メス x 12

Phase 2: 複数センサー例の解説を追加する

- [ ] 複数接続時のコードの書き方を説明する (どうやって 2 つのコードをまとめるか説明する。共通部分の変数はどれで、共通しない部分だけどこからもってくるか。ドライバの読み込みの追加が必要であることなど)
- [ ] example の multiple sensor に SH30 + BH1750 を追加する
  - http://chirimen.org/chirimen-raspi3/gc/top/examples/ 現在 ADT7410+Grove light がある

kou029w, 2020-11-20:
こちら別課題として起票した後、本Issueはクローズとしたいです。
> - [ ] example の multiple sensor に ADT7410 + BH1750 と VL53L0X + MPU6050 を追加する
>   - http://chirimen.org/chirimen-raspi3/gc/top/examples/ 現在 ADT7410+Grove light があるがいずれもメインのセンサーではなくなるので

closed 2020-11-20 by kou029w