chirimen-oh/chirimen.org #68: node-web-gpio/i2c について説明するページを作る

dynamis, 2019-12-02:
Node から CHIRIMEN と同じコードを動かすためのモジュールについて説明するページをつくる

- セットアップ
- hello real world
- ブラウザ版との違い
- ブラウザ版のコードの移植手順

https://www.npmjs.com/package/node-web-gpio
https://www.npmjs.com/package/node-web-i2c

CHIRIMEN Drivers https://github.com/chirimen-oh/chirimen-drivers
ドキュメント https://chirimen.org/chirimen-drivers/ - こちらに書く情報との振り分けはお任せ。一般的な説明を chirimen-drivers に書いて tutorial には初心者向けというのが望ましそうだが。

kou029w, 2019-12-02:
chirimen.org/chirimen-drivers にはどちらかというとそれぞれのドライバーのAPIのリファレンスや新たに対応するデバイスを増やしたいときのための説明書で、対してプロトタイピング教育の講習の中で使うことを目的にした文書は tutorial.chirimen.org に集約されてあるほうが望ましいでしょうね

kou029w, 2020-02-06:
ObnizのクイックスタートページやNode.jsから使うのページが参考になるかも:
- https://obniz.io/ja/lessons/quickstart
- https://obniz.io/ja/lessons/server_side/lessons_nodejs

kou029w, 2020-02-21:
https://tutorial.chirimen.org/raspi/nodejs にチュートリアルページ作成

closed 2020-02-21 by kou029w