chirimen-oh/chirimen.org #75: CHIRIMEN with micro:bit をフルセットチュートリアル化＆同時に全体の見直し

dynamis, 2020-06-17:
現在 CHIRIMEN with micro:bit は CHIRIMEN for RasPi 学習者向けの追加的なものという位置づけで簡便なガイドと Example だけを提供している:

https://chirimen.org/chirimen-micro-bit/

集まっての講習会・イベントなどが難しい昨今、RasPi ではなく安価に入手可能な microbit (+ブレークアウトボード + 電池ボックス) だけで遠隔学習をし易くするため、RPi 向けのチュートリアルを読まず最初から micro:bit 用のチュートリアルだけ読めば済むようにしたい。

そのためには現 Rpi チュートリアルをベースにガッツリ書き直しが必要。全部見直すことになるなら、この際以下の issue を合わせて実行したい:

- [ ] codesandbox ベースに移行 #48
- [ ] コーディングルール更新 #54 
  - [ ] 実行タイミング制御で onload 使わないで defer にする
  - [ ] script defer に関する説明を追加
  - [ ] 関数名の変更 mainFunction から main に
- [ ] 変数名を具体化 #58
  - [ ] 変数名を意味あるものに
  - [ ] const 使う
  - [ ] 1day 講習資料に var, let, const の説明を追加する
- [ ] No More Grove (Grove 編に分離) #61 
- [ ] ADT7410/SHT30 両対応に切り替える #60

dynamis, 2020-06-17:
検討事項:
- Rpi 向けと micoro:bit 向けページの位置関係
  - 独立ディレクトリ・ページのままにするか、統合するか
  - 統合する場合、対象ボードを選ぶと表示が切り替わる Firefox Support の OS 別記述切替のような機能を UI としてもたせるか、共通ファイルから Jekyll で複数ディレクトリに切り分け出力する形にするか
- 開閉式サイドバーの目次を持つテーマに切り替えるか
  - そもそも Jekyll を止める選択もあるんだけど、GitHub Pages との親和性からすると Liquid ベースでかける 11ty が成熟するのを待って移行するのがこのコミュニティ的には良い気もする
- チュートリアルの 1 ページがやたら長いのをもっと細かく分割したりするか？
  - 知ってる人はスルーしたくなる内容は小さな別ページに分けるか、折りたたみ式表示のコラム的なモノにして一見短いページに見せる (その場合でも元ファイルは別 md に分けて include させるなら何度でも埋め込める) か
- jsbin と csb
  - microbit なら PC で編集だから全面的に CSB で平気
  - Rpi も 4 ならまぁ平気だし jsbin とか忘れて CSB に全面移行するか？
- git/github 入門
  - これは別途作成してリンクなり埋め込みなり

kou029w, 2020-06-19:
- [ ] ESModule にしていく
  - 理由: scriptタグによるインポートはCSB環境で強調表示が効かない (ESLintが警告を出す) ので。

hiyohiyo, 2020-07-24:
極力シンプルなのが良いのでCSBへの全面移行に賛成です。
チュートリアルの作成・検証にも携わらせてください。Raspi版との差分はかなりありますが、まずはスタートアップガイドをベースにLチカの周期を変更するところからでしょうか？

satakagi, 2020-07-28:
micro:bitの場合、内蔵デバイスを使う例がありますが、これをどのように位置づけるか？

汎用性(micro:bit以外の環境でも使いまわせる)とか標準性(webGPIO,I2C(これらも標準化には遠いがそれでもそれを意識して使用を作っているかどうかは大きい))とかがあまり感じられるものでないので、副次的な位置づけなことに異論はないと思う。

ただ、micro:bit以外に何も使わず動作することは、はじめの一歩として動作を確認してもらえる　という点で価値が感じられる。(CHIRIMEN with micro:bitは、 CHIRIMEN for RPiと比べてWebBTで接続するという点がハマりやすいポイントがある）

章立てとしては、　CHIRIMEN woth micro:bitの動作確認　という第一章のひとつ前のところで内蔵デバイス使ったチュートリアル入れるのが良いでしょうかね。

dynamis, 2020-07-28:
> 極力シンプルなのが良いのでCSBへの全面移行に賛成です。

micro:bit (PC) や RasPi4 以降であれば CSB (VS Code) 重たい問題もあまりなくなりますし、基本は CSB で、JSBin や JSFiddle は紹介だけ Appendix 的に残すか、完全に消して RasPi3 で non CSB したい人は古いアーカイブサイトを見てくださいとするか (Examples ページも (github pages ではスナップショットビルド残せないので Netlify でアーカイブを作る) ですかね。

> チュートリアルの作成・検証にも携わらせてください。Raspi版との差分はかなりありますが、まずはスタートアップガイドをベースにLチカの周期を変更するところからでしょうか？

ありがとうございます＆こちらが滞っていて申し訳ないです。仰るとおり既存のを両対応にかき分けです。
下記の構成案で進めるなら micro:bit 用 hello real world 部分は raspi との依存関係無しで独立で書けるので、そこをどんな形でも md と画像で書いていって頂けたりすると嬉しかったりします。

> micro:bitの場合、内蔵デバイスを使う例がありますが、これをどのように位置づけるか？
> ... snip ...
> そこで、章立てとして、　第０章？：CHIRIMEN with micro:bitの動作確認　という第一章のひとつ前のところで内蔵デバイス使ったチュートリアル入れるのが良いでしょうかね。

現状 Hello Real World、基礎編、応用編、発展編とありますが、ボード毎の導入、共通のチュートリアル、作品例などの大きく三つに分離し、ボード毎の導入の中に Hello Real World も microbit 内蔵デバイスも入れてはどうでしょう。

### チュートリアル再構成案
パスは新サイト内のパス、URL はベースの現コンテンツ

#### ボード毎のコンテンツ:

- /microbit - CHIRIMEN with microbit について
    - /microbit/hellorealworld- Hello Real World (micro:bit で L チカ) https://chirimen.org/chirimen-micro-bit/guidebooks/startup.html
    - /microbit/sensors 内蔵センサーを使ってみる！ https://chirimen.org/chirimen-micro-bit/examples/
- /raspi - CHIRIMEN for Raspberry Pi について
    - /raspi/hellorealworld - Raspberry Pi で L チカ！ https://tutorial.chirimen.org/raspi/hellorealworld + https://tutorial.chirimen.org/raspi/section0
    - /raspi/howto - Raspberry Pi の使い方、CHIRIMEN 環境
      - ファイルの開き方、編集方法、gc ディレクトリなど CHIRIMEN イメージ固有の部分の説明
    - /raspi/node - Node CHIRIMEN (Node で動かす) https://tutorial.chirimen.org/raspi/nodejs

#### チュートリアル

- /tutorial - チュートリアル
  - 基礎編
    - ここからは原則ボード共通の md で画像、配線図などの差し換え、ボード固有の説明があればかき分け・出力分けする
    - /tutorial/section1 - 1. LED やボタンを使ってみよう (GPIO の使い方) https://tutorial.chirimen.org/raspi/section1
    - /tutorial/section2 - 2. センサーを使ってみよう (I2C の使い方) https://tutorial.chirimen.org/raspi/section2
    - 基礎編にも Grove 編あるといいけどそこまで手が回らない
  - 応用編
    - 基礎編と同じ方針
    - /tutorial/section3 - 3. I2C の使い方 https://tutorial.chirimen.org/raspi/section3
      - /tutorial/grove (TBD) - 3. I2C の使い方 (Grove 編) https://tutorial.chirimen.org/raspi/grove
    - /tutorial/section4 - 4. GPIO/I2C のまとめ https://tutorial.chirimen.org/raspi/section4
  - 発展編
    - /tutorial/communication (TBD) - 複数デバイス間で通信する https://chirimen.org/chirimen-micro-bit/examples/ 
    - /tutorial/cloud-notification (TBD) クラウドサービスでデータのグラフ化と通知 (TBD)
      - IFFFT あるいは machinist とか influxdb とか使って閾値通知とかするのは無償で比較的容易
      - IFFFT の Gmail とか使ってメール通知とかしたくなるが IFFFT に webhook するのが意外と面倒だった記憶がある...
    - その他よくあるユースケースに対応するものがあれば (セクション番号ではなくタイトルで) 追加していく場所に

#### 作品例
- /showcase (TBD)
  - /showcase/playbulb PLAYBULB (LED 電球) を使う https://tutorial.chirimen.org/raspi/section5
  - /showcase/arduino-steppingmotor 6. ステッピングモーターを制御する https://tutorial.chirimen.org/raspi/section6
  - 従来の発展編部分は独立性が高いというかあまり汎用的・シンプルなものではなく、作例みたいなものになってるのでチュートリアルは以下ではなく作例集みたいなものに分離したい。

#### サポートデバイス
- /parts - 対応パーツリスト https://tutorial.chirimen.org/raspi/partslist
  - 現 example 配下のサンプルコードなどを集約。raspi/microbit で別のサンプルページになってるものも同じページから。

パーツリストなどは md で表を書くのではなく yml 編集してその結果を生成する方向にした方が良さそうだなと思ってます。

こういうのと同じような感じで:
https://github.com/gatsbyjs/gatsby/blob/master/docs/sites.yml
https://www.gatsbyjs.org/showcase/

### システム＆テーマ
サイトのシステムとしては既存の Jekyll 実装案はいろいろごにょごにょしたのですが結局それは捨てて、モダンな JAMSTACK の代表格 GatsbyJS を使ったサイト想定に切り替えています。チュートリアルではないトップページなどは React の jsx コードを書くがチュートリアルの執筆に関しては md 書くだけというのは基本変わりません。

ベースのテーマとしては色々確認してみたのですが何だかんだ GatsbyJS 公式サイトチュートリアルの実装が最高に思えるのでそちらをフォーク (gastby 全体の monorepo の一部になってるなどそのままフォークして使うのはしにくいのは諦める):
https://www.gatsbyjs.org/tutorial/
https://github.com/gatsbyjs/gatsby/blob/master/docs/tutorial/

次点としてはシンプルで真っ当な starter が https://github.com/hasura/gatsby-gitbook-starter でしたがちょっと物足りない

satakagi, 2020-07-28:
オリジナルのraspiのreadme.mdをベースに、micro:bit用に書き換えをしてみました。

---
lang: ja
permalink: /microbit/
---

## CHIRIMEN with micro:bit チュートリアル

CHIRIMEN with micro:bit を用いた IoT システムプロトタイピングスキル習得のためのチュートリアル資料です。

JavaScript プログラミング未経験者は、ページ下部で紹介している資料・サイトも参照しつつご覧ください。

## 準備編

micro:bitをwebアプリでコントロールするための準備と最初の動作確認を行います

- [0. Hello web micro:bit](hellomicrobit.md)

動作確認を兼ねて、micro:bitの内蔵センサーやLED等を操作できます。

## デバイス操作～基礎編

シンプルな GPIO 入出力や I2C センサーの操作方法を学びましょう。

- [1. L チカしてみよう (初めての GPIO)](GPIO_starter.md)
  - [WebGPIO API](http://browserobo.github.io/WebGPIO) を使って定期的に LED を点滅するサンプルを動かしてみます。
- [2. GPIO の使い方](GPIO_basic.md)
  - マウスクリックで操作するブラウザ画面のボタンと物理スイッチ (タクトスイッチ) の両方で LED やモーターを制御するサンプルを通じて GPIO の基本を学びます。
- [3. センサーを使ってみよう (初めての I2C)](I2C_starter.md)
  - 温度センサーの値をドライバーを使う場合と [WebI2C API](http://browserobo.github.io/WebI2C) を直接操作する場合の 2 パターンで読み取ることで、I2C デバイス操作の基本を学びます。

## デバイス操作～応用編

いろいろな I2C デバイスを繋いだり組み合わせたりすることで IoT プロトタイピングを体験しましょう。

- [4. I2C の使い方](I2C_basic.md)
  - いろいろな I2C デバイスを接続したり、複数の I2C デバイスの同時操作について学びます。
  - [Grove 編](grove.md)では、Grove を利用したI2C デバイスの同時操作について学べます。
- [5. GPIO/I2C のまとめ](section4.md)
  - これまでの総括として GPIO と I2C の両方を組み合わせて動かしてみます。

ここまでできたら Example 集を見ていろいろなデバイスを試したり自分で好きなものを作ってみましょう。

## IoT編

インターネットを介してデバイスをリモートで操作したりセンシングしたりする、Internet of Things (IoT)の基礎を体験しましょう。

- [6. リモートから温度を測定してみよう](IoT_sensing.md)
  - micro:bit内蔵温度センサを使って遠隔から温度を測定してみます。
- [7. モーターを回してしてみよう](IoT_actuate.md)
  - micro:bitのGPIOに繋いだモータを遠隔からコントロールします。

## 発展編
ここまでの例では飽き足らない人のための発展的な使い方を紹介します。興味のあるものがあればお試しください。
**注意: 学校や講習会などで参加者全員で学習する場合は応用編までで十分です。**
- PCなしでIoTデバイスをつくる
  - [CHIRIMEN for Raspberry PIを使う](../raspi/readme.md)
  - [WebGPIO API や WebI2C API を Node.js から使う](../raspi/nodejs.md)

## 対応デバイスと Example 集

CHIRIMEN with micro:bit 対応デバイスのドライバ・回路図・サンプルコードをセットにした Example 集を用意しています。

[対応デバイスのリスト](partslist.md) : 下記サンプルへのリンク付き

- [GPIO Example](https://chirimen.org/chirimen-micro-bit/examples/#gpio)
- [I2C Example](https://chirimen.org/chirimen-micro-bit/examples/#i2c)
- [Remote Control Example](https://chirimen.org/chirimen-micro-bit/examples/#リモートコントロール-relayserverjsを使います)
- [拡張 GPIO Example](https://chirimen.org/chirimen-micro-bit/examples/#拡張gpio)
- [内蔵デバイス Example](https://chirimen.org/chirimen-micro-bit/examples/#内蔵デバイス)

いずれのExamplesも、元のソースコードは [こちら](https://github.com/chirimen-oh/chirimen-micro-bit/tree/master/examples) に収録されています。

## Appendix, Tips, FAQ, Links etc...

JavaScript 未経験者・初心者向けの資料集や、良くある質問、Tips やテクニック、デバッグとトラブルシューティングのページを用意しています。

- [JavaScript 初学者向け資料集](/js/)
  - JavaScript 1 Day 講習資料、JavaScript 本格入門書、チートシートなどはこちら
- [良くある質問](../raspi/faq.md)
- [Tips・テクニック集](../raspi/tips.md)
- [デバッグ・トラブルシューティング](../raspi/debug.md)
- [講師向けページ](../raspi/teacher.md)

<div class="hide-on-production">
  

  ## Latest Version / 最新版

  Latest version of this document is hosted on https://tutorial.chirimen.org/microbit/

  このサイトの最新オンライン版は https://tutorial.chirimen.org/microbit/ でご覧頂けます
</div>

dynamis, 2020-07-29:
> オリジナルの/raspi/readme.mdをベースに、micro:bit用に書き換えをしてみました。

おぉ、ページの md が issue コメントに書かれるパターン。ありがとうございます。

ちなみに、ざっくりこんな雰囲気で作って行けそうかなと[テストしているもの](https://github.com/chirimen-oh/chirimen.org-gatsby) netlify にもデプロイしておいてみるとこんな感じです。

https://chirimen-org-gatsby.netlify.app/tutorial/

テーマとして使えそうか、振る舞いはどんなものか確認段階なので色々途中です。
- トップや /docs, /blog などは放置でまだまだ [Gatsby 公式サイト](https://www.gatsbyjs.org/) の名残が一杯
- チュートリアルの目次もちゃんとまだ書き直してない
- example やもっと複雑な作例 (発展編にあった様なもの) は https://chirimen-org-gatsby.netlify.app/showcase/ に入れようかなと思っているがデザイン変更などまだまだ
- 画像が欠落して見えるのはレスポンシブ画像生成・変換は従来 cloudinary 使う様にしていたが gatsby ビルド時の画像生成に切り替わるため md 書き直しする必要があるから
- microbit/raspi 共通化・分離生成部分 (肝心なところ) まだ入れてない
- i18n サポートは不完全だったものの対応は後回し

このテーマについてのコメント
- サイトナビゲーション、サイドバーでのチュートリアル全体の目次、ページ上部または右側 (画面幅次第) にページ内目次とナビゲーションが綺麗なデザインで充実してるのは良い
- light/dark モード切替がスマートで良いなぁ。。。(笑)

近日、何とかまとめてこうすれば編集できると分かる＆確定させる様にします。。。

kou029w, 2020-07-29:
> チュートリアル再構成案
> https://github.com/chirimen-oh/tutorials/issues/75#issuecomment-664727783

コンテンツの構成に関してまったく異論ないです :+1: 
どのgitリポジトリか分からないですがいずれにせよ /docs 以下が Jekyll のサイトでいうルートディレクトリに相当するという感じで、それぞれパスに対応するようにMarkdownファイルを配置していくということですかね?

satakagi, 2020-07-29:
今、codesandboxでCHIRIMEN with micro:bitのexamplesを動かそうとしてみてハマりました。
クロスオリジンiframe関係のエラー？
https://codesandbox.io/s/github/chirimen-oh/chirimen-micro-bit/tree/master/examples/GPIO0?module=main.js

kou029w, 2020-07-29:
[iframeの中ではWeb Bluetooth APIにアクセスすることができません](https://w3c.github.io/webappsec-feature-policy/)ので新しいタブで開き直すなどする必要あります > @satakagi 

例: https://github.com/chirimen-oh/chirimen-micro-bit/blob/ca645c26078fa5fb1a2f8df82e919828ea6f9e89/polyfill/demo/main.js

satakagi, 2020-07-29:
codesandboxなら、これを押すわけですね！　ちゃんと説明に書かないと。
[image]

hiyohiyo, 2020-07-29:
こんなご時世ではございますが、近々札幌に帰省することにしたので、子供たちとも試してみたいと思います。
LEDなどを別途使うのも重要ですが、内蔵の温度センサーや傾きセンサーなどが使えるだけでも面白いので、「第０章？：CHIRIMEN with micro:bitの動作確認」は、アリだと思います。物理的なパーツが何もなくてもとりあえず色々試せますからね。

satakagi, 2020-07-29:
今、↑の話を含めて　初期設定・動作確認の章　素案作成中です！

satakagi, 2020-07-29:
https://github.com/chirimen-oh/tutorials/tree/master/microbit
https://github.com/chirimen-oh/tutorials/blob/master/microbit/hello_microbit.md

dynamis, 2020-07-29:
> /docs 以下が Jekyll のサイトでいうルートディレクトリに相当するという感じで、それぞれパスに対応するようにMarkdownファイルを配置していくということですかね?

はい。そんなイメージです。
チュートリアルその他でデザインがある程度シンプル＆固定のものは Markdown (実際には Markdown ではなく MDX) で記述して、そうではない特殊な部分は React (JSX) で記述するという構成です。

Gatsbyjs サイトからフォークするより素の Docusaurus v2 から始めた方が良いのではと言う突っ込みを受けてそちらもお試し中 (こちらになる可能性が今のところ高い) です:
https://github.com/chirimen-oh/chirimen.org-docusaurus
https://chirimen-org-docusaurus.netlify.app/ 

ひとまず今のまま jekyll の markdown でガンガン編集していってくだされば何処かでまとめて or 少しずつ移行作業を行います。

> codesandbox全然よく理解できてないんですが、この新しいウィンドウ(タブ)でページが開けるということは、テンポラリに？このwebAppsがcodesandboxのサービスでだれでも見られるように公開されるということなんですね。　確かに別のPCで、この開いたwindowのURL(ランダムな文字列が入ったもの)を開いてみたらちゃんと見られた！

はい。netlify などがコミットや PR に対してガンガン deploy していくのと同様に CSB もファイル編集して保存する度にそれをガンガン deploy してくれており、それを単独で開くのがその別ウィンドウ表示、エディタないでのプレビューはそれを iframe 読み込みしているだけになります。

> 今、↑のcodesandboxの話を含めて　初期設定・動作確認の章　素案作成中です！
> https://github.com/chirimen-oh/tutorials/tree/master/microbit
> https://github.com/chirimen-oh/tutorials/blob/master/microbit/hello_microbit.md

私が遅くて手が回ってないところをカバー頂きありがとうございます！！！
そのまま tutorial repo で編集し続けて頂ければ、前述の通り適当なタイミングで新デザインサイト側に移植作業します。

その際に  (可能な範囲で) 気に留めて頂けると嬉しい点は以下の通りです:

- 画像は縮小せず大きなままで (今まで通り)
- cloudinary を使うのは廃止する (ビルド時に各サイズの画像生成処理までしちゃいます) ので [現在の編集方法ガイド](https://tutorial.chirimen.org/feedback) こちらに書いている「画像の埋め込み」については無視して普通の markdown での画像埋め込みをしてください。
- markdown 中に HTML を埋め込む必要がある場合 br や img などを開きタグだけで `<br>` 等と書かず `<br/>` など整形式になる様に書いてください。Markdown に HTML じゃなく JSX 的タグを埋め込む MDX として処理するための制限事項です。
- ページ間リンクは相対リンクで。具体的パスは URL は移植時に適当に変えるのであまり気にしないでください。パスを変えてリンク切れが起きてもビルド時チェックで検知出来るのでそれで見つけて直します。

dynamis, 2020-07-29:
あると嬉しい (作りたい) 新規独立ページ:

- CSB の使い方や仕組み (@kou029w さんお願いします)
  - VSCode のフロントエンドをブラウザで動かしていて使い勝手はほぼ同様
  - 既存サンプルのフォークの仕方 (編集して ctrl+s するだけでもできてしまう)
  - ctrl+s など基本ショートカット、保存されているファイル、されていないファイルの見分け方
  - iframe 無いに表示されるが別ウィンドウで独立、そのままデプロイされててスマホや他の PC でもアクセス可能な公開サイトを即席で作ることになる (公開しちゃ不味いことを保存しない様に！)
  - eslint や prettier がデフォルト有効でそれの警告などの見方
    - 設定何処にあるのか、設定の説明があるサイトへの誘導
  - アカウント無しでも使えるがアカウントアリなら PC 間で編集を引き継いだり制約が色々無くなるからログインして使おう
  - github 連携なしでも使えるが github 連携で repo の編集が出来る
  - 他にも色々あるけど WIMC 的な場面で役に立つ知識の範囲でカバーする
- microbit 標準搭載機能の制限事項 (@kou029w お願いします)
  - 標準機能の使い方ページに追記でも独立でも良い
  ｰ https://tinkering-mbit.github.io/microbit-workshop-guide/devices.html などに書いていった様な注意事項をこちらでも書いておく
- webgpio/webi2c の使い方と API リファレンス
  - 仕様書は流石に初心者にはハードル高いが長ーいチュートリアルの中での説明しかないので簡潔な説明と API リファレンスが欲しい。内容や書きっぷりはMDN 的なイメージ
  - 特に仕様に書いてない拡張はどのボード・実装でしか使えないかというラベル付きで
  - 良くあるエラーと原因 (i2c 競合してるよ！とか i2c デバイス繋がってない or ポートやアドレス違うよ！とか)
    - raspi 向けのコードをうっかり microbit で使おうとしたときに良くあるものも意識して
- microbit の主なブレークアウトボードの紹介とピン配列や付加機能の違い
  - 国内入試し易い pin:bit と、弊社でそれなりに在庫を用意したのが m:bit, iobit を中心にその他も随時
  - ブレッドボードと一緒に使う
    - **リファレンスは pin:bit** https://www.sengoku.co.jp/mod/sgk_cart/detail.php?code=EEHD-5AWW
    - **pin:bit とほぼ同じだが少し違い安価な m:bit** https://ja.aliexpress.com/item/4001118934669.html
    - SparkFun https://www.sengoku.co.jp/mod/sgk_cart/detail.php?code=EEHD-56JS
  - ブレッドボード無しで使う
    - **安価な IOBIT** https://ja.aliexpress.com/item/4000087499661.html
    - 高機能版 IOBIT v2 https://ja.aliexpress.com/item/32890247131.html
    - ブザーなどとピンの競合・があるのに

kou029w, 2020-07-31:
- [CodeSandboxガイド](https://hackmd.io/@kou029w/HyKj-hybD) … 作成中
- microbit 標準搭載機能の制限事項 … 未着手

satakagi, 2020-07-31:
https://github.com/chirimen-oh/tutorials/blob/master/microbit/GPIO_starter.md
本格的にRPiとコンテンツの大部分が一致しているはずのページを　RPiページをベースに　ひとまず何も考えずに編集してみました。
たしかに大部分は共通　ただ、細かい部分で違うところがチマチマある・・

satakagi, 2020-08-03:
https://github.com/chirimen-oh/tutorials/blob/master/microbit/GPIO_basic.md
も作ってみました。githubpagesが更新されないのは気のせいでしょうか？

kou029w, 2020-08-04:
> githubpagesが更新されないのは気のせいでしょうか？

[Deploy Log](https://app.netlify.com/sites/tutorial-chirimen-org/deploys/5f28ae5be16e240007a95ea2)によるとmicrobit/GPIO_starter.mdからのrelative_include examples/section0/s0.htmlなどの部分で参照に失敗していたようです。一旦その部分をコメントアウトしてデプロイしました。s0.htmlを正しく配置するなどしてもらえれば直ると思いますのでご確認ください。 > @satakagi

dynamis, 2020-08-06:
> CodeSandboxガイド … 作成中

ありがとうございます！ぱっと見ではあとこのあたりの記載でしょうか？

- 既存サンプルのフォークの仕方 (編集して ctrl+s するだけでもできてしまう)
- ctrl+s など基本ショートカット、保存されているファイル、されていないファイルの見分け方
- iframe 無いに表示されるが別ウィンドウで独立、そのままデプロイされててスマホや他の PC でもアクセス可能な公開サイトを即席で作ることになる (公開しちゃ不味いことを保存しない様に！)
- eslint や prettier がデフォルト有効でそれの警告などの見方
- 設定何処にあるのか、設定の説明があるサイトへの誘導

WIMC リモート開催文脈としては (あるいはプログラミングの遠隔講義一般にだけど) 次の 2 つは独立ページで説明が欲しい:

- CSB 上で既存 sandbox からのフォーク、自分の github repo として保存するまで
  - CSB で他の人の (CHIRIMEN コミュニティで用意したテンプレの) sandox を開く
  - CSB 書き換えて編集、保存
  - CSB 上で github ログイン、リポジトリ名を指定して Create Repository する
  - CSB 上で続けて普通に編集、コミット、プッシュ
- CSB 上で共同編集する (チーム作業 or 遠隔で助けて貰う) 手順
  - 開く
  - Live タブでリンクを取得、共有して他に人に開いて貰う
    - 参加する側も github ログインしている必要あり
  - Classroom モードと共同編集モード (Everyone can edit) があることを説明

dynamis, 2020-08-06:
note:
Algolia OSS https://www.algolia.com/for-open-source/
Crowdin OSS https://crowdin.com/page/open-source-project-setup-request 
Crowdin Academic https://crowdin.com/page/academic-license-project-setup-request

satakagi, 2020-08-07:
chirimen micro:bitのI2C examplesを倍増させ、tutorialsのSection4の準備ができました。（Section4で紹介しているデバイスの半分がexamplesになかった）

hiyohiyo, 2020-09-14:
遅ればせながらチュートリアルを確認中です。
https://tutorial.chirimen.org/microbit/gpio_basic

中盤にある以下の部分が解決できないようです。これは、最終的に変換されるものなのでしょうか？
{% cloudinary imgs/section1/LEDOnOff.gif alt="LED On/Offをブラウザ画面上のボタンクリックで実施" %}

それとも、https://github.com/chirimen-oh/tutorials/tree/master/raspi/imgs/section1/LEDOnOff.gif にリンクするように修正した方が良いですか？

kou029w, 2020-09-25:
@hiyohiyo はい、そちらはJekyll(とそのプラグイン)によって、適切なリンクに変換されるので、そのままで構いません。([Feedbackの画像の埋め込み](https://tutorial.chirimen.org/feedback#section-2:~:text=%E7%94%BB%E5%83%8F%E3%81%AE%E5%9F%8B%E3%82%81%E8%BE%BC%E3%81%BF,%E5%A4%96%E9%83%A8%E3%82%B5%E3%82%A4%E3%83%88%E7%94%BB%E5%83%8F%E5%9F%8B%E3%82%81%E8%BE%BC%E3%81%BF%E3%81%AF%E3%81%97%E3%81%AA%E3%81%84%E3%81%A7%E3%81%8F%E3%81%A0%E3%81%95%E3%81%84)を参照)

kou029w, 2020-11-20:
> CHIRIMEN with micro:bit をフルセットチュートリアル化＆同時に全体の見直し #75

前者は完了済みですかね?
後者の全体の見直しはIssueを分けたほうが良さそうに思います。いかがでしょうか。