chirimen-oh/chirimen.org #79: アーカイブサイトが Google の誤判断で偽装サイト扱いされている

dynamis, 2020-11-16:
2018 年度のアーカイブサイトが Chrome さんに偽サイト扱いされてしまってる

https://webiot-2018--tutorial-chirimen-org.netlify.com/
> アクセスしようとしたサイトは偽装サイトのようです。不正なユーザーが URL のごく一部を一目ではわからないように改変してサイトを偽装する場合があります。

「無視」を選ぶか Firefox 等であればアクセスが可能だが、再審査要求して解除されるのが望ましい:
https://developers.google.com/web/fundamentals/security/hacked/request_review

dynamis, 2020-11-22:
2019 年度のアーカイブもだ...
https://webiot-2019--tutorial-chirimen-org.netlify.app/raspi/

kou029w, 2020-11-22:
解決してませんか?

dynamis, 2020-11-23:
お。解決したっぽいか。では、再発したら reopen ということで。

dynamis, 2020-12-05:
やっぱまた起きてる。誰かが警告無視したら一定時間みんな OK になるがまた暫くしたらアウトとかだろうか。
アクセス数の少ないアーカイブサイトがニセサイト扱いされやすいのではないか疑惑。。。

なお、メインドメイン化 #88 に伴い netlify の default subdomain 名変更したのでブランチデプロイ URL も次のように変わった (https://github.com/chirimen-oh/chirimen.org/commit/c605274d4b0d1e96e365c988e4d3a9e20415066d) :

https://webiot-2019--chirimen-org.netlify.app/raspi/
https://webiot-2018--chirimen-org.netlify.app/raspi3/ja/