chirimen-oh/chirimen.org #85: node 版をよりアピールする

dynamis, 2020-11-24:
現在 Node 版は RasPi チュートリアルの発展編の一項目扱いで掲載中

https://tutorial.chirimen.org/raspi/
https://tutorial.chirimen.org/raspi/nodejs

ピン配列など RasPi 用の実装になっていて汎用化できてないところもあるので raspi 配下という感じになってるが、需要もあるしより目立つように配置したい。

- /raspi/ 配下に残す場合は専用の段落で説明する
- いっそ /nodejs/ にしてしまって raspi ベースで説明するに過ぎません扱いにするか

dynamis, 2021-02-06:
- トップなどの説明で「ブラウザ」とだけ書いているのを「ブラウザや Node.js」で同じコードで書けるという説明に更新した。
- about ページで Node.js からも使えることをアピール、リンク

dynamis, 2021-06-10:
pizero を headless で使う場合のドキュメントも書いてくれたものを一緒に整理する
https://tutorial.chirimen.org/pizero/pizeronodejs

dynamis, 2021-09-13:
note: https://tutorial.chirimen.org/pizero/pizeronodejs ⇒ https://tutorial.chirimen.org/pizero/ (readme.md) という形で新しいものになっており、 readme 側が完成 (？) となった段階で pizeronodejs の方は削除、リダイレクト設定とする。

dynamis, 2021-09-13:
- 外部ページ参照部分も整理して取り込みたい
  - ブラウザから設定するツール : https://qiita.com/mascii/items/a43d71572e1919e56398
  - PiZero を USB OTG 接続 : https://github.com/webdino/pizero-workshop/blob/20190528/docs/Setup.md
    - 特にこちらはドライバーのインストールをさせてるが公開先が消えているなどあるのでそのままだと辛い

> CHIRIMEN ブラウザー版との差異

環境がブラウザ/node いずれかを navigator オブジェクト有無などで判断してグローバル変数 requestGPIOAccess などを作ってしまうことでユニバーサルに振る舞うコードテンプレを用意して既存サンプルも差し換えができると共通化できて良いかも知れない？

dynamis, 2021-09-13:
トップページからの誘導はした: https://github.com/chirimen-oh/chirimen.org/commit/9befc1e07923f59156c871402e8f2cda0362695e

- [ ] サイト共通のヘッダからの誘導
- [ ] /about
- [ ] /raspi からも誘導

node 版は pizero 配下をメインにしていく
- raspi/nodejs 配下を整理継続
- /raspi/nodejs も /pizero 配下に pizero 向けの記述メイン (model b 向けも残すか分けるかなにか) で移しても良い

gurezo, 2024-04-09:
https://github.com/chirimen-oh/chirimen.org/issues/88 が終わり次第