chirimen-oh/chirimen.org #88: チュートリアルサイトをコミュニティトップに差し替える

dynamis, 2020-12-03:
20201117 ミーティングにて、現状の chirimen.org をアーカイブにした上でこのリポジトリでメンテしているチュートリアルサイトを chirimen.org に設定することになった。

- [ ] chirimen.org を現 tutorial.chirimen.org (このリポジトリ) に変更
  - [ ] netlify 側の設定を変更
  - [ ] chirimen.org の DNS 設定を netlify 向けに
- [ ] tutorial.chirimen.org を chirimen.org へのリダイレクトサイトに
  - [ ] 新規リポジトリ作成
  - [ ] netlify の設定
  - [ ] tutorial.chirimen.org の DNS 設定を更新 (もしかしたら netlify 側の設定だけで DNS は更新しなくて良いかも)
- [ ] 現 chirimen.org のアーカイブサイトを netlify でデプロイ
- [ ] 現 chirimen.org アーカイブサイトにチュートリアルの about 辺りで説明、リンクをする
- [ ] サイトのトップをよりコミュニティトップらしく
  - お知らせ的なものの記載欄を追加していく
  - などなど

あとは例えば次のようなことを順次出来れば (この issue の範囲外) 
- node 版の位置づけをより分かりやすく #85
- partslist のパス変更 #86 と整理
- 電子工作一般の説明ページ分離 #87
- チュートリアル以外の共通コンテンツの配置・紹介を整理
  - JS 初心者向け、github ガイド、codesandbox ガイドなど

dynamis, 2020-12-03:
本リポジトリを netlify 上で (現状の) tutorial.chirimen.org に加えて chirimen.org の aliase としても設定した。
DNS 側の設定はこのようにすると chirimen.org でチュートリアルサイトが表示されるようになる:

[image]

gurezo, 2023-05-09:
@dynamis @dadaa 
DNS の管理や費用で、どうなっていましたっけ？

dadaa, 2023-05-09:
アタシがもってますよ。次の更新は今年の 9/30 です。DNSの設定もこちらでできそうです。

gurezo, 2023-05-09:
@dadaa cc @elie-j  @dynamis 
今日ミーティングで、ドメイン費用とかどんな状況だっけという話が出まして
今後管理や費用も含めて、どうしようか相談したいという話になりました。

dadaa, 2023-05-10:
ほい。どうしましょ？

gurezo, 2023-05-13:
@dadaa cc @elie-j @dynamis
すみません、今気が付きました。
DNS・ドメインの管理やドメイン費用負担 を @dadaa さん個人から
Web DINOの移譲したいのですが、如何でしょうか？

dadaa, 2023-05-13:
りょ。

gurezo, 2023-05-14:
@dadaa 
ありがとございます！
@elie-j @dynamis  以降の移譲の段取りをお任せしても良いでしょうか？

dynamis, 2023-06-26:
@gurezo スミマセンこちらのやり取り見落としてました。
弊社でのドメイン費用負担(保有管理)に切り替えるなら DNS 移管作業ですね。

@dadaa さんの方でドメイン移管の AuthCode を発行して DM か何かでご連絡頂けないでしょうか。

移転時にアクセス不能にならないよう DNS 設定引き継ぎなどで困ったら追加質問あるかもですが、多分 AuthCode さえあれば後はこちらで作業できそう (移転後は Cloudflare DNS を使う＆既存設定を自動で読む機能があるので多分問題ない想定)。

弊社側のレジストリの説明 https://www.value-domain.com/userguide/manual/transfer
移転後は DNS 管理権限などはコミュニティの中の何名かにばらまいて私がボトルネックにならないようにさせてください。

gurezo, 2023-08-24:
@dynamis 
今の状況を教えてください 🙇 
何をしたら良いか把握したいです

chibitaki, 2024-04-04:
@dadaa さんに再度対応していただくように連絡しました。

dadaa, 2024-04-04:
おつかれさまですー
AuthCode 確認のところで移管先を訪ねられているので、送ってくださいませー
Slack でもメールででも。

dynamis, 2024-04-04:
移管先として何の情報が必要ですか？ GMO (value-domain) のアカウントに移転したいのですがこちらの ID か何かの入力を要求されてます？

[image]

dadaa, 2024-04-04:
value-domain だけわかれば大丈夫です。ありがとさま。あとで slack で送りますわ。

chibitaki, 2024-04-04:
@dadaa slack で送ってくださったとのこと、私にも入っていますか？
フォローするので、私にも送ってください。いろいろすみません。
よろしくお願いいたします。

dadaa, 2024-04-05:
おはようございますー
@chibitaki @dynamis Japanese Mozilla community group の slack で良いんですよね？ さきほどお二人にメッセージ送ったのですけれど、もしかして届いていないのかしら？

chibitaki, 2024-04-05:
@dadaa ありがとうございます！届きました！

dynamis, 2024-04-05:
どうやら元のドメインの contact のメールアドレス確認をするようで、whois proxy を解除するのが最低限必要なようですので、まずは whois proxy 解除頂けますか @dadaa 

[image]

まずはその状態で手続き進められるか試したいのですが、更に、どうやら記載のメールアドレスがこちらの登録アドレス (webmaster at webdino.org) と一致しなければ進められない可能性が高そうです:

https://www.value-domain.com/transfer.php?action=notes
> ドメイン登録者連絡先（registrant contact）として自分のメールアドレスが登録されていない、また、そのメールアドレスでメールを受信できない場合。

普通は他のドメイン管理なども一緒にしていて、登録アドレスまるごと変えるわけにはいかないでしょうし。

汎用jpドメインなら譲渡手続きはちゃんとシステムとして存在するし個人で使ってる [muu-muu だと譲渡手続きが明確に定められてる](https://support.muumuu-domain.com/hc/ja/articles/360046369153-%E3%83%89%E3%83%A1%E3%82%A4%E3%83%B3%E3%81%AE%E8%AD%B2%E6%B8%A1%E6%96%B9%E6%B3%95)のだけど、value-domain の方はマニュアル探しても譲渡手続きの説明が見当たらず。

そちらのドメイン管理サービス (とレジストリ) の方には説明などありますか？

dadaa, 2024-04-05:
@dynamis 解除してみました。どう？

dynamis, 2024-04-05:
@dadaa 進行できました。そちらでの承認手続き操作をお願いします:

[image]

dynamis, 2024-04-08:
移転完了しました！

[image]
[image]

新しく chirimen 専用の cloudflare アカウントを作成、DNS セットアップ (既存レコードのインポート) して NS 
alex.ns.cloudflare.com, athena.ns.cloudflare.com ⇒
ricardo.ns.cloudflare.com, rosalyn.ns.cloudflare.com
へと切り替え。

dynamis, 2024-04-08:
自動インポートしてくれてると思った DNS 設定が全然既存の設定じゃなかったのかリダイレクトループ？になっていたので一旦 dns サーバを旧サーバに戻してから設定調査。

dynamis, 2024-04-08:
取りあえず後回しにしようとしていたが、旧 DNS 設定依存とかはいつまで稼働できるか分からないし、デプロイから新しい Cloudflare Pages 環境に揃えた:

旧 chirimen.org (github pages, 古いもの) の cloudflare pages デプロイ
https://chirimen-oh.pages.dev/
デフォルトでビルド通らなかったので取りあえず build system version 2 (ruby3.2.2) ⇒ 1 (ruby 2.7.1) にしたら通った。これを archive.chirimen.org に割り当ててリンクなどを張っていく想定

tutorial.chirimen.org の cloudflare pages デプロイ
https://chirimen-org.pages.dev/

_redirect rule 非互換ありそうだなと思ったらでプロログに次の warnings:

```
23:10:13.306 | Parsed 7 valid redirect rules.
-- | --
23:10:13.307 | Found invalid redirect lines:
23:10:13.307 | - #5: https://tutorial-chirimen-org.netlify.com/*   https://tutorial.chirimen.org/:splat 301!
23:10:13.307 | Only relative URLs are allowed. Skipping absolute URL https://tutorial-chirimen-org.netlify.com/*.
23:10:13.307 | - #15: /raspi/examples/polyfill/*  https://chirimen.org/chirimen-raspi3/gc/polyfill/:splat  200
23:10:13.307 | Proxy (200) redirects can only point to relative paths. Got https://chirimen.org/chirimen-raspi3/gc/polyfill/:splat
23:10:13.308 | - #16: /raspi/examples/drivers/*   https://chirimen.org/chirimen-raspi3/gc/drivers/:splat  200
23:10:13.308 | Proxy (200) redirects can only point to relative paths. Got https://chirimen.org/chirimen-raspi3/gc/drivers/:splat
23:10:13.308 | - #18: /raspi/examples/section3/*   https://chirimen.org/chirimen-raspi3/gc/i2c/:splat  200
23:10:13.308 | Proxy (200) redirects can only point to relative paths. Got https://chirimen.org/chirimen-raspi3/gc/i2c/:splat
23:10:13.308 | Info: the redirect rule /raspi/partslist → 302 /partslist could be made more performant by bringing it above any lines with splats or placeholders.
23:10:13.308 | Info: the redirect rule /raspi3/appendix0 → 302 /js/async could be made more performant by bringing it above any lines with splats or placeholders.
23:10:13.308 | Info: the redirect rule /raspi/appendix0 → 302 /js/async could be made more performant by bringing it above any lines with splats or placeholders.
```

absolute URL ができないってのはそもそも netlify 向けのハックなので削除すれば良い
Proxy (200) redirects ダメよってのは修正が必要かも

CF Pages へのカスタムドメイン設定前:
[image]

カスタム設定に伴う変更
[image]

設定後 DNS 設定
[image]

取りあえずここで namesever 再度 switch

gurezo, 2024-04-09:
- tutorial サイトが未達？
- 他は移管完了

dynamis, 2024-04-09:
tutorial.chirimen.org にアクセスできませんか。手元の複数経路と監視サービスは問題ないのですが、一部のDNSキャッシュに駄目なの残ってたりしたかな。。。？継続して駄目なら調べるのでお知らせください。

404の挙動やアクセスログからもnetlifyからcloudflareデプロイ側に切り替わってるなと確認していたつもりなのですが。

暫く切り替え安定が確認できたら当初のarchiveに移す＆メインサイトをtutorialに差し替えるとかの作業に進めたいなと思ってます。

kou029w, 2024-04-09:
https://chirimen.org/PiZeroWebSerialConsole/
https://chirimen.org/chirimen-drivers/
など、/リポジトリ名/ が現在 404 Not Found となってしまっているようです

dynamis, 2024-04-10:
tutorial サイトに切り替えるには chirimen/gc などのサブディレクトリで使われているものがあったので単純差し換えはまだ出来ないな、と思っていたところでした。

> など、/リポジトリ名/ が現在 404 Not Found となってしまっているようです

そうですよね。ディレクトリベースの gh-pages も結構利用されていたのでそれらは CF にする場合は個別デプロイ / リダイレクト設定をする必要がありますね。影響範囲調査と対応方針検討を終えるまで一旦 gh-pages ([DNS 設定はこちらを参照](https://docs.github.com/ja/pages/configuring-a-custom-domain-for-your-github-pages-site/managing-a-custom-domain-for-your-github-pages-site)) に戻しました:

CNAME @ chirimen-oh.pages.dev
  ⇒
A @ 185.199.108.153
A @ 185.199.109.153
A @ 185.199.110.153
A @ 185.199.111.153

アクセス出来るようになったが証明書が (chirimen.org 向けではなく github.io 向けとされて) エラーになってるのでカスタムドメインの設定を続けて実施。

org での Verified domains に chirimen.org を追加 ([出来るように DNS TXT レコードの設定と verify 処理を実施](https://docs.github.com/ja/pages/configuring-a-custom-domain-for-your-github-pages-site/verifying-your-custom-domain-for-github-pages))
https://github.com/organizations/chirimen-oh/settings/pages

しかし個別リポジトリ側の pages 設定にカスタムドメインの設定がでてこない。。。

https://docs.github.com/ja/pages/configuring-a-custom-domain-for-your-github-pages-site/managing-a-custom-domain-for-your-github-pages-site
[image]

[image]

https://docs.github.com/ja/pages/configuring-a-custom-domain-for-your-github-pages-site/troubleshooting-custom-domains-and-github-pages#https-errors
> カスタムドメインを設定した後、サイトが HTTPS 経由で利用可能になるには最長 1 時間かかります。

ということだからだろうか。少し待ちながら継続調査

dynamis, 2024-04-10:
tutorial サイトのためのリポジトリの設定を見ていた。DNS Check in Progress とあるので素直に 1 時間待つ:

[image]

dynamis, 2024-04-10:
gh-pages を使ってきているリポジトリと設定画面 URL をリストアップ:

- http://chirimen.org/
  - https://github.com/chirimen-oh/chirimen-oh.github.io/settings/pages
- http://chirimen.org/chirimen-micro-bit/
  - https://github.com/chirimen-oh/chirimen-micro-bit/settings/pages
- https://chirimen.org/chirimen-TY51822r3/
  - https://github.com/chirimen-oh/chirimen-TY51822r3/settings/pages
- http://chirimen.org/chirimen-drivers/
  - https://github.com/chirimen-oh/chirimen-drivers/settings/pages
- http://chirimen.org/node-web-gpio/
  - https://github.com/chirimen-oh/node-web-gpio/settings/pages
- http://chirimen.org/node-web-i2c/
  - https://github.com/chirimen-oh/node-web-i2c/settings/pages
- http://chirimen.org/PiZeroWebSerialConsole/
  - https://github.com/chirimen-oh/PiZeroWebSerialConsole/settings/pages
- http://chirimen.org/remote-connection/
  - https://github.com/chirimen-oh/remote-connection/settings/pages
- http://chirimen.org/technical-book8/
  - https://github.com/chirimen-oh/technical-book8/settings/pages
- http://chirimen.org/webGPIO-etc-on-microbit-via-webBluetooth/
  - https://github.com/chirimen-oh/webGPIO-etc-on-microbit-via-webBluetooth/settings/pages
- http://chirimen.org/chirimen.org-multiple-languages-test/
  - https://github.com/chirimen-oh/chirimen.org-multiple-languages-test/settings/pages
  - これは試験リポジトリなので維持する必要なし

CF pages にデプロイしてサブディレクトリ gh-pages からはリダイレクト設定を書くという対応をすれば良いかと思ったのだが、ちょっと多いな。。。

しかし gh-pages 継続利用しつつ tutorial サイトを apex ドメインに切り替える場合 netlify/cloudflare では使えていた/使っている機能 (_redired など) が使えなくし、jekyll 縛りがあって docusaurus とか他のドキュメントサイトに使いやすいフレームワークへの移行も不可能。

ちょっと多いけど頑張って片っ端から CF pages (chirimen.org サブディレクトリではない gh-pages でも良い) にデプロイしていってサブディレクトリからはリダイレクトする設定を書いた上で tutorial サイトを apex ドメインに切り替えるという手順を踏む以外には利用中の全機能を維持/今後の更なる柔軟性を確保する手段はなさそうかな。。。

kou029w, 2024-04-10:
デプロイについて:

> jekyll 縛りがあって

この条件はGitHub Actionsを使用して回避できるかとは思いますよ

参考リンク:

- [カスタム GitHub Actions ワークフローによる公開](https://docs.github.com/ja/pages/getting-started-with-github-pages/configuring-a-publishing-source-for-your-github-pages-site#%E3%82%AB%E3%82%B9%E3%82%BF%E3%83%A0-github-actions-%E3%83%AF%E3%83%BC%E3%82%AF%E3%83%95%E3%83%AD%E3%83%BC%E3%81%AB%E3%82%88%E3%82%8B%E5%85%AC%E9%96%8B)
- [GitHub Pages でのカスタム ワークフローの使用 - GitHub Docs](https://docs.github.com/ja/pages/getting-started-with-github-pages/using-custom-workflows-with-github-pages)

dynamis, 2024-04-10:
> この条件はGitHub Actionsを使用して回避できるかとは思いますよ

そうそう、そうでした！Actions 利用制限のある org とかだったりもしませんものね。
リダイレクトは使えなくて構わない (あるいは meta でのリダイレクトで許容できる) と判断できれば tutorial のデプロイを gh-pages で行う形で apex ドメインに使えるようにするのもありですね。

dynamis, 2024-04-10:
gh-pages の DNS check が完了し chirimen.org での HTTPS 証明書エラーが出なくなり、ひとまず障害は一通り解消できただろうか。

もう少し現状で問題が残ってないか様子を見た上で、gh-pages に寄せるべきか CF pages に寄せるべきか迷う/判断し、apex ドメインを tutorial サイトに差し替える作業を進める。

www.chirimen.org の設定がないよって gh-pages からは指摘されてるが以前からそれは使ってなかったはず。
試しに tutorial サイトを www.chirimen.org 側に cf pages デプロイした状態にしておいてみる / 試験環境として使ってみる (後日 bulk redirect 設定で chirimen.org に飛ばすことを想定)

closed 2025-09-22 by kou029w