chirimen-oh/chirimen.org #101: examples が browser/node で分かれているのを纏められないか

dynamis, 2021-06-10:
browser/node 版それぞれを使う人にとっては必要十分な情報がこちらにある

- browser: https://chirimen.org/chirimen/gc/top/examples/
- node: http://chirimen.org/chirimen-drivers/examples/

但し、少し気になるところがある

- node 版の説明ペーじ https://tutorial.chirimen.org/raspi/nodejs  からデバイス説明ページに辿り着きにくい
- 同じような情報が 2 箇所にあり、ハードの説明とかリンクの追加とかを sync できないのが勿体ない

また、Remote 版のサーバは node 版も用意すれば自由に組み合わせて使えるようになったりするし、browser/node で分けずに CHIRIMEN のデバイスの説明とそこからそれぞれの利用例をリンクするような形にした方が良さそうに思う。

kou029w, 2021-06-10:
https://github.com/chirimen-oh/remote-connection/blob/master/js/RelayServer.js 読むと下記のAPIがNode.jsではデフォルトでglobalThisに存在しないと思うので対処が必要そうですね:

- WebSocket
- TextEncoder
- Scaledrone

dynamis, 2021-06-10:
Scaledrone はライブラリ実装されてるヤツで、こいつの node 向けライブラリがあるのかという問題になりますね
`<script src='https://cdn.scaledrone.com/scaledrone-lite.min.js'></script>`
https://github.com/chirimen-oh/remote-connection#scaledrone

dynamis, 2021-06-10:
TextEncoder って MDN 的には Node 11.0 からサポート的に書いてるけど使えないのかな。
使えないなら使えないでここに書いてある Polifill も一緒に突っ込んでも良いけど。
https://developer.mozilla.org/ja/docs/Web/API/TextEncoder

kou029w, 2021-06-10:
https://github.com/chirimen-oh/chirimen/blob/aa74b01e86c5a58acf14739f5980f84b8deac0eb/setup.sh#L95-L97

今 setup.sh 見ると v12 なので問題無さそうですね :ok_man: > TextEncoder

dynamis, 2021-06-10:
WebSocket はライブラリに入れ替えるか universal なモジュールに差し替える程度にするか？
https://www.npmjs.com/package/universal-websocket-client

どうやっても良いけど確かにちょっと手当は必要

dynamis, 2021-06-10:
> Scaledrone はサービス側がライブラリとして提供してるヤツで、こいつの node 向けライブラリがあるのかという問題になりますね

オフィシャルドキュメント https://www.scaledrone.com/docs/api-clients/javascript は browser 向けのものしか案内してくれてないけど node module これ使えば良い感じかな。こいつが両対応で動けば分けずに済むが...
https://www.npmjs.com/package/scaledrone-node