chirimen-oh/chirimen.org #108: チュートリアルサイトのパーツ画像変換バグ確認

gurezo, 2022-03-14:
@dynamis 
https://tutorial.chirimen.org
上記に表示する画像の変換して配置する仕組みをコメントで教えて下さい
メンテしたいです。

kou029w, 2022-03-14:
問題の背景:  Cloudinary での変換後の画像が表示されない問題があったとのこと (@satakagi さんより)

> 上記に表示する画像の変換して配置する仕組みをコメントで教えて下さい

参考文献: フィードバック > 画像の埋め込み https://tutorial.chirimen.org/feedback#section-2

dynamis, 2022-03-15:
cloudinary で画像が表示されない！というケースとしてはオリジナルサイズの画像に初回アクセス/ロードする前に変換後の画像へのアクセスをして (一定時間) ブロックされる場合があります。

変換元の画像を最初に読み込み、サーバに保存させた上でそれを次回からそのまま or 変換してキャッシュを送るという仕組みかつ、存在しない(初回アクセスされていない)画像へのアクセスをすると不正アクセスと判断して一定時間ブロックする仕組みもあるようで。

それ以外は特に使っていて困ったケースがあった記憶はありません。
具体的に何処の画像に問題があるか分かればもう少し。

余談としては、この辺りの問題に何度か出くわした後は、ビルドが重たいのを我慢できるケースは Web フレームワーク側でビルドしてサイズ別の画像ファイルを生成しちゃう Static Site Generator 使う方が罠がなくて良い (かも知れない) と思いました。

satakagi, 2022-03-15:
https://github.com/chirimen-oh/chirimen.org/blob/master/raspi/imgs/section1/s.png
が更新した画像（タクトスイッチを普通の4ピンタイプに for https://github.com/chirimen-oh/chirimen.org/issues/92 ）
で、https://tutorial.chirimen.org/raspi/section1#b-
で使われている実態配線図なのです。github上のpngを置き換えただけなので、置き換わってくれると期待したのですが、今も古いまま(2pinタクトスイッチな図)な感じ

satakagi, 2022-03-18:
https://tutorial.chirimen.org/raspi/imgs/section1/s.png
なのに、

https://res.cloudinary.com/chirimen/image/fetch/c_limit,f_auto,q_auto,w_1000/https://tutorial.chirimen.org/raspi/imgs/section1/s.png
となっている（古いまま）ということですね。

もうcloudinary使うのは止めますかね？ @dynamis

gurezo, 2022-09-06:
一旦close

closed 2022-09-06 by gurezo