chirimen-oh/chirimen.org #136: downlaod.chirimen.org にアクセス出来ない

dynamis, 2024-05-30:
[レジストリ＆DNS 切り替え](https://github.com/chirimen-oh/chirimen.org/issues/88#issuecomment-2046384562)のタイミングで恐らく何かミスをしてしまって download.chirimen.org にアクセス出来ない状態になっていると思われる。

ホストされていたのは恐らく Google Drive の共有ディレクトリの CHIRIMEN uploader ではなかろうか

https://drive.google.com/drive/folders/1s8Aa0XCrJWxiHFn_UFBCHThOSgj6meCp

とも思ったがこれらはログイン無しで勝手にアクセスできる状態ではない。何か別の Web Server であったかもしれない。
要確認＆要対応

kou029w, 2024-05-30:
> 何か別の Web Server

おそらく techorus03.mozilla.jp に向けてあったようです

2021-08-12-chirimen.zip イメージのダウンロード方法の具体例:

- https://drive.google.com/file/d/1hAFjwYOflz8sO_B760oHwenmehtKrunq/view?usp=drive_link (暫定)

```
$ curl -fsSL -H Host:download.chirimen.org techorus03.mozilla.jp/2021-08-12-chirimen.zip -O
```

dynamis, 2024-08-08:
before
[image]

after:
[image]

にて https://download.chirimen.org/ で image リストにアクセス可能になったことを確認 (該当サーバの方がまた別途メンテナンス/差し換えを必要としているがそれはまた別の話)。

合わせて Cloudflare の CHIRIMEN チームの super admin に watanabe@webdino.org を招待しました
(他にも権限付与すべきメンバーはご連絡いただくか渡邉さんから展開していただければ)

kou029w, 2024-08-08:
ありがとうございます
アクセスできるようになったことを確認しました
ホスティングサーバーを変える件は別途とし本件はクローズします〜

closed 2024-08-08 by kou029w