chirimen-oh/chirimen.org #137: PiZero版チュートリアルのリニューアル

Aritaka-hub, 2024-06-06:
@satakagi @kou029w 

PiZero版のチュートリアルサイトを WIMC+ の各地のメンターや学校関係者の方も見る（使用する）機会が増えてきた事もあり、より使い勝手を良くするための UI 変更をメインとしたリニューアルの作業をしたいと思います。

今の構想はサイドメニューを付けて各情報のアクセスを容易にする変更を考えていて、内容変更までは考えていませんが、リニューアル作業をしてみて色々とご意見いただけるとありがたいです。プレビュー版など出来たらまた相談させてください。

dynamis, 2024-06-06:
> 今の構想はサイドメニューを付けて各情報のアクセスを容易にする変更を考えていて

サイドメニューというのはサイト全体の共通？ pi zero チュートリアルページ https://tutorial.chirimen.org/pizero/ 単体に閉じたもの？それとも pizero 専用のサイドメニューのあるページの右に既存の色んなページを埋め込むフレーム的なもの？

目次にあたるものが折りたたみで再度 (両端) に表示できるという形であれば [docusaurus](https://docusaurus.io/docs/next/installation) などを使うだけで自動的にチュートリアル全体の目次 (現在ページがそのどの部分か) と各ページ内での目次が共に左右に折りたたみで表示できるようになったりします(例は docusaurus 自身のもの https://docusaurus.io/docs/next/installation とか [react](https://react.dev/learn), [svelte](https://svelte.dev/docs/introduction) その他フレームワークの docs サイト) が、そういうシステム的なものを使ったりしますか？

pi3+ の方は導入ページ https://tutorial.chirimen.org/raspi/ があって配下に内容毎に分かれるという構成を取ることで必要な情報にアクセスし安くということをしていました。また、パーツリスト(およびそこから続く examples) などのようなものはサイトトップの共通ヘッダに入れる事でケアしようという設計でした。

docusaurus 的なものでチュートリアル全体/ページ内のナビゲーションを改善するのは ([ドメイン移転の作業](https://github.com/chirimen-oh/chirimen.org/issues/88)先として止まってしまってますが) サイトのシステム変更とか構成変更で[過去にも何度か議論してました](https://github.com/chirimen-oh/chirimen.org/issues/75#issuecomment-665898227)がそういう話とは別ですか？

dynamis, 2024-06-06:
個人的にはいまのチュートリアルサイトで教えるときにやりにくいと思っているのは

- 自分が使ってるデバイスのための情報だけが欲しい
  - pi3+ (browser) と pizero (node) で実装が違うのにサイト内で混在しており初めてさんに分かりにくい
- 足りない情報がある
  - pi3+ と pizero (あるいは microbit) などの一部のデバイス向けだけに用意されている情報がある
- 頻繁にアクセスするページ/ツールへのリンク集的なものが (特にpizero) ない
  - hellorealworld、デバイスリスト、exmaples、[serial console](https://chirimen.org/PiZeroWebSerialConsole/PiZeroWebSerialConsole.html)、i2cdetect コマンドの使い方、端末間連携 (IoT編)、常駐化 ... あたりはチュートリアルを流すのと関係なくさっと試すとき/継続的に使う時に必要だからそれらが存在することに気付くように/辿りやすくしたい
- デバイス毎の情報が分散している
  - partslist に書いている情報と pi3+ 専用の example に書いている情報が分散している
- その他細かな問題
  - リンク切れ、コードの不統一、レイアウトやスマホ対応不足、パーツの説明不足(写真が違う/i2cアドレスが違う、配線が違う/足りない)、ドライバの説明/機能不足

これは多分、Pizero 配下のページだけ直せば済むもの、ページ分割と共にサイトのヘッダやサイドバー(サイドメニュー)などの導入/更新でできるもの、複数デバイス共通ページの再整理まで必要なものと別れるというイメージですが、どの辺りを解消するためにどうやるのかというイメージが分かる形で説明なりラフスケッチなりプレビューなり見られると嬉しいです。

kou029w, 2024-07-19:
本Issueではチュートリアル全体 (#75) の改修とは分けて、pi zero チュートリアルページの改修を想定します。
具体的には、/pizero/ 配下を部分的に mdBook https://github.com/rust-lang/mdBook で、あるいは Docusaurus で書き出す、部分的でインクリメンタルなリリースを考えています。

Pizero 配下のページでは以下が課題という認識です:

> - 頻繁にアクセスするページ/ツールへのリンク集的なものが (特にpizero) ない

どうやるのかというイメージとしては @Aritaka-hub としてはこのように分割して、目次が共に左右に折りたたみで表示できるようにしていく、という想定です (分割した Markdown のコードがオンライン上に見当たらなかったのでお手数ですが後で共有してもらえれば > @Aritaka-hub):

> - [1. 概要](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_1.html)
>     - [1.1. CHIRIMEN について](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_1-1.html)
>         - [1.1.1. CHIRIMEN コミュニティ](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_1-1-1.html)
>     - [1.2. Raspberry Pi について](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_1-2.html)
>     - [1.3. JavaScript の基礎](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_1-3.html)
>     - [1.4. 良く使う情報へのリンク](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_1-4.html)
> - [2. 機材準備と Wi-Fi への接続](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_2.html)
>     - [2.1. ステップ0（機材準備）](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_2-1.html)
>     - [2.2. ステップ1（ターミナル接続）](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_2-2.html)
>         - [2.2.1. Note. CLI操作について](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_2-2-1.html)
>         - [2.2.2. Note. pi4 の利用方法](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_2-2-2.html)
>     - [2.3. ステップ2（Wi-Fi設定）](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_2-3.html)
> - [3. Hello Real World（Lチカを実行する）](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_3.html)
>     - [3.1. 配線](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_3-1.html)
>     - [3.2. プログラムを書く](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_3-2.html)
>     - [3.3. 実行する](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_3-3.html)
>     - [3.4. コードを読む](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_3-4.html)
> - [4. GPIOを試す](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_4.html)
>     - [4.1. GPIOを理解する](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_4-1.html)
>     - [4.2. GPIOを出力](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_4-2.html)
>     - [4.3. GPIOを入力（onchange）](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_4-3.html)
>         - [4.3.1. onchange コードを読む](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_4-3-1.html)
>     - [4.4. GPIOを入力（ポーリング）](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_4-4.html)
>         - [4.4.1. ポーリングコードを読む](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_4-4-1.html)
>     - [4.5. GPIOセンサーを複数同時に使う](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_4-5.html)
>         - [4.5.1. Lチカのコードを書き換える](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_4-5-1.html)
> - [5. I2Cデバイスを試す](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_5.html)
>     - [5.1. I2Cを理解する](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_5-1.html)
>     - [5.2. SHT30編](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_5-2.html)
>         - [5.2.1. I2Cセンサー(SHT30)の認識を確認する](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_5-2-1.html)
>         - [5.2.2. SHT30 のコード取得と実行](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_5-2-2.html)
>         - [5.2.3. SHT30 のコードを読む](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_5-2-3.html)
>         - [5.2.4. SHT30 のコードの詳細解説](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_5-2-4.html)
>     - [5.3. ADT7410編](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_5-3.html)
>     - [5.4. GPIO と I2Cセンサーを組み合わせる](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_5-4.html)
> - [6. IoTを試す](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_6.html)
>     - [6.1. 遠隔LEDコントロール](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_6-1.html)
>     - [6.2. PiZero サンプルコードの実行](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_6-2.html)
>     - [6.3. PC サンプルコードの実行](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_6-3.html)
>     - [6.4. 自分専用チャンネルで制御](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_6-4.html)
>     - [6.5. PiZero 側のコードを読む](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_6-5.html)
>     - [6.6. PC 側のコードを読む](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_6-6.html)
> - [7. 常駐プログラム化する](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_7.html)
> - [8. 他のいろいろなデバイスを試してみる](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_8.html)
>     - [8.1. 応用センサーキットの使い方](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_8-1.html)
>         - [8.1.1. 単体で動作確認できるセンサー](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_8-1-1.html)
>         - [8.1.2. 人感センサー](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_8-1-2.html)
[... cut ...]
>         - [10.6.6. セキュリティ](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_10-6-6.html)
>         - [10.6.7. Node.jsでの利用](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_10-6-7.html)
>         - [10.6.8. Webhooks](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_10-6-8.html)
>         - [10.6.9. IoTクラウドサービス](https://aritaka-hub.github.io/kataria.github.io/pizero/book/chapter_10-6-9.html)

https://aritaka-hub.github.io/kataria.github.io/pizero/book/

gurezo, 2024-07-19:
@kou029w 
横から失礼します。
電子書籍の[構成](https://github.com/chirimen-oh/chirimen.org/pull/130)をこちらにコメントさせて頂きます。
参考になれば、幸いです。

Aritaka-hub, 2024-07-19:
@kou029w  コメントありがとうございました。
マークダウンのコードですが GitHub にアップ出来ていませんでした。

https://github.com/Aritaka-hub/kataria.github.io/tree/main/docs/pizero/src
こちらに一式アップ出来ました。色々不慣れですみません。

現在のチュートリアルで、特に遠隔講義を行う時にサイトのどこが分からないかを共有しやすくする目的で分割を行っています。また CHIRIMEN Generic の内容も統合して情報集約を行い初学者にも読みやすくなればと思い構成してみました。
まだ情報量の精査など十分ではないかもですが、方向性の共有が出来れば幸いです。

@gurezo  情報共有ありがとうございます、こちらも作業の参考にさせて頂きます。

kou029w, 2024-08-07:
@gurezo 構成参考にさせていただきます! ありがとうございます!
@Aritaka-hub も共有ありがとうございます!

現状としては

- プレビュー https://mdbook.www-chirimen-org.pages.dev/pizero/ を閲覧可能
- ブランチ [mdbook](https://github.com/chirimen-oh/chirimen.org/tree/mdbook) https://github.com/chirimen-oh/chirimen.org/tree/mdbook にコードを配置済み (これをmasterブランチに取り込めば [chirimen.org](http://chirimen.org/) には反映可能な状態)

です
このあとマージしてみます
お手数おかけしますが問題無さそうかどうか https://tutorial.chirimen.org/pizero/ をご確認ください〜 > 皆様

closed 2024-08-09 by kou029w