chirimen-oh/chirimen.org #138: Chorome で 2回目以降の Monaco エディターの内容が表示されない場合がある

Aritaka-hub, 2024-07-02:
Chorome の現在のバージョンで、WebSerialConsole から Monaco エディターで編集を行おうとした際に、最初の 1回目は問題なくコードが表示されますが、エディターウィンドウを閉じた後すぐに編集で開くと内容が表示されなくなりました。
- Windows10 Pro 22H2 （2024/7/1 最新のセキュリティパッチ適応済み）
- Chorome バージョン: 126.0.6478.127（Official Build） （64 ビット）

白表示された Monaco エディタのウィンドウのスクショと、DevTool のエラースクショを貼っておきます。

しばらく時間をおいて、編集よりコードを開くと表示される事もありますが、ウィンドウを閉じて直ぐに開こうとするとほぼ 100% 白画面で表示されません。

MS edge ではこの問題は起きず、いつ編集を実行してもコードは問題なく表示されました。

[image]

[image]

kou029w, 2024-07-12:
@Aritaka-hub 
フィードバック感謝です
変更点についてですが、HTMLシンタックスエラーを見つけましたので修正しました (ただ、直接これが問題の原因であるかどうかはワタシは判断できていません)
再度ご確認いただけると幸いです
変更箇所 https://github.com/chirimen-oh/PiZeroWebSerialConsole/commit/aafa0d68a6890bb78f3f5a5dfa000a696e9f61ed

kou029w, 2024-08-22:
ワタシの手元では試していて再現しなったようなのでクローズします
もし不具合再現するようであればお手数ですがコメントいただけると幸いです
(その場合はリオープンします)

closed 2024-08-22 by kou029w